HumanDynamics/openbadge #108: Clarifications before fabricating badges

patcon, 2018-10-12:
First off, thank you so so much for making this repo public and keeping it up-to-date. I've had such energizing conversations with people about working with the artifacts of your project.

### Context

I'm just getting ready to order some badges in unofficial collaboration with @shuyanglin of [PDIS](https://pdis.nat.gov.tw/en/), the Taiwanese government's digital innovation department created by [Digital Minister Audrey Tang](https://qz.com/767298/taiwans-new-digital-minister-is-a-transgender-software-programmer-who-wants-to-make-government-more-open/).

We're currently getting ready to order a pilot batch of 15 sociometric badges and 3 BLE beacons. We'll later order 85 more, for use at a civic hackathon and other loose experiments. The pilot and larger scale experiment will likely have two separate goals:

1. to take certain online processes into physical space (pilot), and
2. to help the civic tech community better understand itself (larger experiment)

### Disclaimer

I don't have hardware experience myself. I'm eagerly collaborating with people like @kevinphy of Taipei Hackerspace. Having said that, I accept full responsibility for any stupid questions I ask in this issue :)

### Questions

#### Circuit Board Fabrication

1. Are the newest files exports always the ones you'd recommend? (I will be assuming "yes" with my following links and questions, but please advise if incorrect. I can imagine considerations of stability vs bleeding edge, but I'm not clear how to account for that.)
2. Will we need to order/build our own "programmer"? Could we manage without? Is this [all the details](https://github.com/HumanDynamics/openbadge/tree/master/eagle/nRF_badge/Badge_03v2-Prog) for doing that?
3. What is the difference between [`Badge_03_v6`](https://github.com/HumanDynamics/openbadge/tree/master/eagle/nRF_badge/Badge_03_v6) and [`Badge_03_v6_LIS2DH12_breakout`](https://github.com/HumanDynamics/openbadge/tree/master/eagle/nRF_badge/Badge_03_v6_LIS2DH12_breakout)? Which is recommended? These are the facts as I see them in relation to this question:
    - :mag: All the export dirs in v6 have BOMs, but v6_LIS2DH12 lacks a BOM
    - :mag: The files were all committed at the same time, so no clues there.
    - :mag: all the BOMs in v6 exports actually have the LIS2DH12 component (3-axis accelerometer) included [(example)](https://github.com/HumanDynamics/openbadge/blob/master/eagle/nRF_badge/Badge_03_v6/export_20171108/badge_03v6_bom.csv)
4. Will `Badge_03v2-Prog` (assuming this is the programmer device), work with more recent v6 badges?

#### Misc.

1. Which BLE beacons would you recommend?

We'll be happy to contribute back what we learn by expanding on the documentation in this repo!

## Note

I emailed David Shrier before arriving in Taiwan, in relation to my hopes of running a sociometric badge experiment. He responded quickly and graciously, and tried to point me in the right direction, but it seemed the direction was away from the badges, which is the primary interest. Apologies for any confusion this second outreach causes :)

OrenLederman, 2018-10-12:
These are all good question.

1. The newest export is the right one to use (we've made ~200 units using this design so far). https://github.com/HumanDynamics/openbadge/tree/master/eagle/nRF_badge/Badge_03_v6/export_20180201 

2. You'll need to make your own programmer. The link you mentioned has the design itself. It has a very simple design - order the board from any PCB maker, and solder the headers yourself (the LED is optional). You'll need a few more parts though. There is more information on that here - https://github.com/HumanDynamics/openbadge/wiki/Open-Badge-V3.0#building-a-programmer (note - you don't have to make the programming frame/rig, just get the pogo-pin pkit from sparkfun)

3. Ignore Badge_03_v6_LIS2DH12_breakout. It was just a test

4. Yes, this programmer will work for all versions of the badge

5. I actually never really used BLE beacons - I always use my badges as location beacons. It gives me more flexibility. also, the new version of the server+hub allows you to program them remotely (which saves time). If you do get BLE beacons, make sure they support the iBeacons protocol. You'll need to manually set their IDs using the program that comes with them. In the past, I used beacons from these companies https://store.radiusnetworks.com/  and https://estimote.com/

A couple of notes about fabrication - the fabrication is straight forward, but different PCBAs have different equipment. In our case, the company had to use small glue dots to keep the passives (capacitors and resistors) in place so they don't "tombstone".

patcon, 2018-10-13:
@OrenLederman, you are my hero. I'll process this new info with our group, and we'll keep this issue updated with progress. (I'll create new issues for any specific code/doc contributions.)

patcon, 2018-10-25:
Any recommendations on whether to even attempt surface-mount soldering on this board?

I'm [reading this article](https://hackaday.com/2016/03/17/so-youre-scared-of-surface-mount/), and trying to decide if these components are small enough that a novice should avoid, even to save a little assembly money. I'm seeing some 0603 size parts, which I think might be too small. Curious your advice, or whether anyone on your team did anything but pick-and-place. Thanks! :)

OrenLederman, 2018-11-08:
Keeping this thread open until we document this somewhere