HumanDynamics/openbadge #109: Host project on community website for easier ordering

patcon, 2018-10-25:
Hi again @OrenLederman!

A few of us have been talking and meeting to work on this. (We keep [notes](https://hackmd.io/s/H1mc0oj9Q#Meetings) fwiw)

We came across some websites that we were hoping to use so that it's simpler for the next group to walk this path. They allow for project sharing, and essentially one-click ordering for others who want to do the same fabrication.

- [PCBWay: Shared Projects](https://www.pcbway.com/project/)
- [seeed studio: Project Hub](https://project.seeedstudio.com/)
- [Elecrow: Cooperated Designers](https://www.elecrow.com/cooperated-designers) (seems more profit-motivated and top-down than others)

*Would you be opposed to openbadge having a presence on community platforms like this?

Here are some comments -- 🔍  facts, ❤️ feelings, and 💡 ideas -- related to this discussion:

_community platforms:_ Meaning community side of PCBWay, seeed studios, etc.

#### General

1. :mag: The openbadge license is MIT, and so we're technically free to do anything.
1. :heart: I feel preference to work in a way that aligns with your expectations and goals.
1. :heart: I feel eager to make it simpler for others to start projects with openbadge.
1. :heart: I feel eager to actively contribute to having a presence on community platforms, so that I can help foster a larger peer community of collaborators.
1. :heart: I feel open to letting your team manage this, if you have strong feelings on it.
1. :mag: The community platforms don't seem to offer accounts for team management.
1. :bulb: We could share an single account.
1. :mag: seeed studio has [some options for teams/contributors](https://project.seeedstudio.com/guidelines) that we could use accordingly.

#### Profit-sharing and incentivization on platforms

1. :mag: The platforms seem to offer financial incentives (paid or discount).
1. :heart: I feel uninterested in being personally enriched by hosting a project on these platforms.
1. :heart: I feel that the incentive could be used to promote something community-minded.
1. :bulb: We could enabled "monetization", but use it to give community projects discounts/credits.
1. :bulb: Alternatively, we could try to disable any monetization on the account.
1. :heart: I feel that monetization should be disabled if it in any way increases the price of regular orders.
1. :mag: I've [tweeted](https://twitter.com/patconnolly/status/1055375819437047808) at PCBWay to discover options for commission.
1. :mag: The community platforms have referral programs, eg [PCBWay's](https://www.pcbway.com/helpcenter/promotions/How_does_the_referral_program_work_.html?search=).
    1. :bulb: We could use the above-mentioned shared account to receive referral bonuses from doc/readme links, and hold it for community uses.

Anyhow, I look forward to hearing your thoughts! Hopefully this is a helpful format for this discussion :)

cc: @kevinphys (pls feel free to add your own thoughts! 😄 )

OrenLederman, 2018-10-25:
Hi guys,

(also adding @jordanreedie  and @MichaelHopfengaertner to the discussion)

I think it's a great idea, and will be happy to help. I was playing with the idea of trying order badges from PCBWay or Seeedstudio so I can make the assembly process more generic and easy for people. We are getting many requests for badges, so it can be a good way for enabling people to make their own. 

Monetization might be a good idea. I'm not planning to get rich because of this project, but this money can help support some more development of the system and benefit the community (for example, designing a badge firmware "burning" tool that is easier to use...).

The only thing is that I currently don't have time to take lead on this (trying to wrap up my PhD this year, and handling a 2mo baby and a 2yo toddler..). But I will help with whatever I can. If there is enough interest, I might be able to pull in some more resources (students) to help.

Oh, also - we should refer to the project as Rhythm Badge (OpenBadge is an official Mozilla project, and they asked us to change the name). At some point i'll rename the repositories...

Also - fun fact - we have a new firmware version in the making (in a fork of the project). A complete re-write that will make it easier for people to contribute to the project. We are almost done testing all the existing functionality, and will soon start testing some new stuff (better protocol, accelerometer support, etc). 

Cheers,
Oren

eskofier, 2018-10-25:
Oren, in case you are too busy, we can build badges for fellow researchers. 
All, it would be important to know the demand for badges. 10, 100, 1000?

OrenLederman, 2018-10-25:
I actually haven't been keeping good track of the requests recently. Typically, people ask for some andges (about 10) to try out , and later on might need more. But making the badges by itself isn't too time consuming for me. Here are the problems I see:

* I can't sell badges since MIT is a non profit
* Ordering badges is about Ricky nowdays because of the shortage in capacitors. It requires fidning alternative capacitors every time and place the order quickly. 
* Badges then need to be tested and programmed
* And then, there's the initial support for using them and basic analysis :)

So by creating a community I'm hoping to made the information needed more available and help people help each other. Hopefully, in a way that scales.

patcon, 2018-10-25:
> Ordering badges is about Ricky nowdays because of the shortage in capacitors.

I _knew_ Ricky was behind it. He is admittedly a bit tricky.

patcon, 2018-10-25:
And @eskofier, might that offer apply to civic technologists who are working on citizen science initiatives? We are part of the [g0v.asia](https://g0v.asia) civic tech community (prounounced "gov-zero"), working loosely with [PDIS](https://pdis.nat.gov.tw/en/), part of the Taiwanese government.

We were looking to do a first run of 15. No pressure. (Email is patrick.c.connolly @ the googles)

OrenLederman, 2018-10-25:
Bah. That's what happens when I try to write quickly on my phone. I miss my
blackberry keyboard...

OrenLederman, 2018-10-25:
It's getting busy in here :-) . 
@patcon - another thing to consider is hosting a simple forums system somewhere to make it easier to ask questions and create discussions. Is this something that Seeed/PCBWay offer as part of their community projects?

patcon, 2018-10-25:
I've been investigating the seeed/pcbway systems for most of the afternoon, and it seems that they're pretty rudimentary. They offer comments and "likes", but no forums per se. I wouldn't want to try to manage a community over there -- just allow ordering and simple interactions. I'll think on other options. I might lean toward github (perhaps another repo) until that fails or gets complaints, just because that seems in flow with the other work flow?

PCBway actually seems better than seeed. PCBway allows you to share a project, and someone can add a PCB to the cart. There is a specific place on the project page to attach a BOM, and I think the idea is that you can then add another line item to the cart by asking for assembly. It's not seamless, but it's not bad.

Another option, though unsure how pricing would look: https://learn.sparkfun.com/tutorials/how-to-sell-your-widget-on-sparkfun#option-2-sparkfun-manufactures-you-get-a-royalty (perk of sparkfun is that they're fully assembled and have a high standard for docs and support)

OrenLederman, 2018-10-25:
I think that github is ok for tracking issues, but sometimes you just want to have a discussion, or ask questions. In particular, people ask me a lot of questions about tips for deployment and data analysis. Saving these questions in a forum instead of emails can be useful :) I'll try to look into that, and see where I can host something like that.

OrenLederman, 2018-10-26:
I'm not a huge fan of synchronous chat for large discussions. I think it's a good tool when you routinely work with other people and when you need to coordinate stuff (and we might need it sometimes).  

What I had in mind is more of a forum system. I'm thinking of the other use cases that I see, which is mostly Q&A and support - manufacturing/fabrication, questions about deployments (base stations and server) and data analysis. Creating topics will allow you to reduce clutter and make it easier to search (think stack overflow). 

I'm not sure what the solution is right now, I'll have to think about it for a bit.

OrenLederman, 2018-10-26:
Also, going back to one of the original questions here - does anyone have a preference for PCBWay or SeeedStudio? I can try and order a small batch of badges from one of them and see if the quality is ok

patcon, 2018-10-27:
PCBWay seems to have a better "project sharing" feature, so I was gearing to order from them. Thanks for asking!

Seeedstudio (as in, their project sharing feature) seems to be more for ppl coloring within the lines, building stuff from polished kits)

patcon, 2018-10-30:
Not ready for primetime yet, but I've set up stubs for project pages, so that others can see what it might look like to host on PCBWay:
https://www.pcbway.com/project/shareproject/Rhythm_Badge.html
https://www.pcbway.com/project/shareproject/Rhythm_Badge_Programmer.html?x=1

kasbah, 2018-10-30:
Hey, creator of Kitspace here. Obviously biased but some advantages I believe Kitspace has over other options:

- It's the only site that lets you purchase the parts required for self-assembly with a single click
- You can easily review part information in the BOM view
- Open source
- Not tied to any specific manufacturer or parts retailers (we currently link to specific ones but will keep adding to these)
- It's a front-end to your existing Git repository, if you change anything on Github it will be updated on Kitspace automatically

Full disclosure: I do currently earn and collect 10% commission on any order referred to PCBWay and Aisler. This has earned me about $70 in one year :slightly_smiling_face: . Once this hits some kind of workable threshold (where we can support the development of the site with it) we will look into passing this on to project creators.

If you are happy to put it up on Kitspace, I am happy to help you out with it. So let me know. Either way, this is a really interesting project. Do you think it could help people with [Prosopagnosia](https://en.wikipedia.org/wiki/Prosopagnosia)?

OrenLederman, 2018-10-30:
Thanks Kaspar (are you monitoring conversations about Kitspace? :)). I'll look into it more, but so far I see two issues:

1. When you place the order through Kitspace is only fills in the PCB info, but not the PCBA
2. A really cool feature that I'm missing is assigning a list of alternative parts. Especially now with the world shortage in capacitors.

kasbah, 2018-10-30:
Haha, no. @patcon @'ed me so I got a notification. 

1. Yes, it's only for PCB info for now. Extracting all the PCBA info and making sure it's correct is much harder. Does PCBWay offer this on their project sharing platform?
2. You can keep adding MPNs per line for alternatives. For the retailer parts (SKU) you have to select one per retailer, should figure out adding alternatives there too. :thinking:

OrenLederman, 2018-11-06:
Thanks @mannokwong , that'll be very helpful

patcon, 2018-11-08:
Been playing around with a service called Octopart, for comparing and ordering from variation distributors. It's a pretty good user experience to sanity-check the parts, and shop for alternatives if things get sold out or go on back-order

Here's the BOM, staying as try as possible to the most recent BOM for v3.6:
https://octopart.com/bom-tool/TAh4HILh

I'll leave that one as is, and create copies for any experimentation with other parts.

We should strive to store out version-controlled BOM in a format that's a simple as possible to import -- I had to do some munging in a spreadsheet. The eagle settings:

- export values-based BOMs that have a column for "Quantity"
- "Designator" for part (comma-separated)
- It would need just one identifier of any type (MFN or specific distributor ID like mouser or digikey)
- "Description" column would be helpful for the sanity check part. It's a description like `RES 6.8K OHM 5% 1/10W 0603` taken from Digikey, and makes it simpler to swap out parts in their interface but still check to be safe.

OrenLederman, 2018-11-14:
Hi Patric,

I suggest moving the BOM recommendations to a separate issue (if you haven't already), just so it's easier to track.

Going back to the forum - I think that a google group is a good solution. I prefer a forum over a "chat", since It allows you to open issues and organizes the information better. 

Project page (and documentation) I should be able to host on our MIT site. I'll check.

mannokwong, 2018-11-28:
[image]
[image]
I have finally received the badges from the manufacturer :D 
I tried Seeedstudio, but you can try other like PCBway. 

And the next step would be the programming part...

patcon, 2018-11-28:
Oh, we received our parts and boards from pcbway this past weekend, using the octopart BOMs above. Will be meeting on Tues to assemble some

patcon, 2018-11-28:
> Project page (and documentation) I should be able to host on our MIT site. I'll check.

Might we instead be able to use GitHub pages or some other platforms where the process to contribute changes is clear and direct?

OrenLederman, 2018-11-28:
@mannokwong - the badge looks good in black :) I take it that you placed the components yourself? For a very quick check, if you put a battery in a turn it on, the red LED should turn on. It'll only validate that the nrf51 chip is ok.

@patcon - let me look into the github pages option. It looks like a good option for a community driven solution. A website would require user management and permissions. So far, it sounds like github pages (for content) + google group (for discussions) is a good option.

OrenLederman, 2018-11-30:
Hi all, I opened a google group for us - https://groups.google.com/forum/#!forum/rhythm-badges 
We should move general discussion and question there.

patcon, 2018-11-30:
Thanks Oren! Did you intend to create it as a Discussion Board? If so, these don't allow forwarding emails:
[image]

OrenLederman, 2018-11-30:
Yup. Following your advise, I activated posting via email (and allowed posting questions too).
Does it look ok now?

patcon, 2018-11-30:
:+1:

And can we remove the need to request access, and just make it immediate to join? If that creates a future issue, we can easily roll back in response.

OrenLederman, 2018-11-30:
good idea. done

mannokwong, 2018-12-03:
[image]

@OrenLederman Yeah, seems it works (the red LED turning on!). But I found that it is so difficult to remove the battery.

OrenLederman, 2018-12-03:
Oh. You can either use a very thin flat screwdriver, or better yet, a metal
sim card removal tool. The later a very cheap on ebay. You stick them on
the lower right corner of the battery holder and just pop the battery out.

patcon, 2018-12-03:
@OrenLederman Would you be in favour of making the wiki editable by all github users? Was going to add your comment somewhere in there.

Everything in wiki is git backed, so even on the remote change of vandalism, it's still a non-issue so long as someone notices. (And if more people care for the wiki, that's more people to notice :) )

OrenLederman, 2018-12-03:
I think it's better to wait until I open a new repo with github pages. It's
safer to have these changes reviewed as pull requests. What do you think?

In the meantime (until I get to this... I'm focusing on some analysis that
need to be finished soon), we can collect these changes as part of the main
README so they are not lost/forgotten.