HumanDynamics/openbadge #111: Discuss making more use of Octoparts' Common Parts Library

patcon, 2018-11-01:
I'm still figuring out how this usually works, but is the common parts library something that was considered while building? Basically, using components from the list helps ensure stable stock and cheapest prices.

https://octopart.com/common-parts-library

https://www.open-electronics.org/find-all-the-electronic-components-of-your-dreams-on-common-parts-library-octopart/

patcon, 2018-11-01:
Tracking research here: https://docs.google.com/spreadsheets/d/1WTSAATEkbLkV2WBjTKi6NyAyRwMRTKHSMc90ohGy0Dc/edit#gid=388279007

OrenLederman, 2018-11-14:
This activity will have to wait, from my side. I'll only make changes when I have more time and when I can run a PCBA batch to validate that it's working. @eskofier is recruiting a student to help with making badges, and we'll ask him to streamline the process