HumanDynamics/openbadge #115: Updating the image of the badge to newest version

bhimasta, 2018-11-20:
Hi @OrenLederman,

Since the last email that I sent to you, I have been able to order the badge through pcbway.com. It will take around one month for the entire process. The notes that organized by @patcon are becoming a huge help to this point.

In the past couple of days, I got some emails from them asking for several issues, such as whether they did the right things on soldering the switch. These questions were probably very easy questions for you.

2. What to do with the extra wielding points?
[image]

2. If this part is soldered correctly?
[image]

3. If this part is soldered correctly?
[image]

Honestly, I thought that giving them enough information such as the schematics, BOM, and centroid, everything will go smoothly. 

As I don't have any experience with this hardware engineering, so answering this questions is complicated for me. 

So the first things that I do as a beginner is to compare the picture of the rhythm badge that I order with the image of the rhythm badge showed in the readme.md, to see if the soldered in the right position and direction. 

However, It seems that the picture of the rhythm badge showed in the readme.md is different with a different position and different looks of the components. Is it the older version? If yes, perhaps, It would be better if you could update the image with the newest version, so people (like me), can easily validate our artifacts through visual.

Thank you very much.

OrenLederman, 2018-11-20:
I've been (very slowly..) looking at pcbway, but simply didn't have time to try and order from them myself :) . Looking at the files they request, they ask for the centroids, but I think that does not include the angle for each component. In most cases it's easy to tell, but not always.

In this case:
#2 - the extra welding points are for an alternative battery holder that we used to use in the past. Cheaper, but not as strong. It's ok to solder on top of it

#4 - both these components (microphone and switch) are fine

#3 - Based on what I see, I think it's ok. If the arrow is pointing at pin 1, then it's correct.

When you communicate with them, can you tell them that your colleagues (me) would like to make similar orders, and ask what would be the best way to provide the placement information? Also, ask them if there is something we can change in the files to make it easier for them.

I'll try to bring my high resolution camera to the office next and take an update photo

OrenLederman, 2018-11-20:
Ummm, at a closer look, I think they messed up some of the resistors - R3, R2, and R8 should be 1k resistors, but I think they placed 100k resistors (probably using the same ones they used for R5, R6, R7).

Can't tell if they messed up the capacitors as well (they aren't marked the same way resisters are).

bhimasta, 2018-11-20:
Yeah, they asked me about the centroid in the beginning and I (beginner) thought that the rotation column gives the info about the angle of each component. Is it not the case?

Sure, I will ask them on would be the best way to provide the placement information and other things that make it easier for them.

Oh, Thank you for also notice the resistor issue, I will also try to follow up this issue with them.

patcon, 2018-11-20:
This is awesome. Thanks for sharing @bhimasta :) moderate resolution photos of new badge versions would be really helpful when we're manually assembling ours (parts in transit now)

bhimasta, 2018-11-20:
@patcon, I would love to join your manually assembling sessions if you don't mind. :D

bhimasta, 2018-11-21:
@OrenLederman 

I just got the replied from them

> As for R2, R8, RC0603JR-071KL 1K- the silkscreen is 01B. If 100K, then silkscreen is 01D.
> So we think we have soldered R2 R8 correctly, what do you think?

> For efficient assembly, we would suggest you offer the following files:
> 1. Centroid file, see attached coordinate file for reference
> 2. Assembly Instruction Drawing - display the polarity of Diode, Audion, LED, pls use “+” ”-”; 
       display the orientation of IC, eg, dot for IC”s Pin1. 
       see attached components position file for reference

>3.3D map file-display the THT parts’location and orientation

>4. **Assembled real picture if you ever made, it will be the best reference.**

[Template_Components position file.pdf](https://github.com/HumanDynamics/openbadge/files/2602482/Template_Components.position.file.pdf)

OrenLederman, 2018-11-21:
>    1. *Assembled real picture if you ever made, it will be the best

I'll copy their recommendation to a new issue.
The resistors are ok. It was hard to see in the image, but indeed the
marking is 01B, with is +-1%, as opposed to the %5 I'm usually using (which
is perfectly fine)

OrenLederman, 2018-11-21:
Here's an temporary updated photo. I'll update the readme + place a copy with the gerber files after I process and clean it up 

[image]

bhimasta, 2018-11-22:
Thank you very much Oren!

closed 2019-01-22 by OrenLederman