HumanDynamics/openbadge #116: [Support] Question from PCBWay on PCB manufacturing

patcon, 2018-11-21:
We've ordered programmer and badge boards from PCBWay. Programmer boards went off without a hitch. Badge boards had a quality assurance question:

> Ref attached screenshot, is it correctly designed as no hole is there for the pointed out pad ?

Attached: 
[image]

patcon, 2018-11-21:
In the meantime, I forwarded @bhimasta's image from the other thread, so hopefully that will answer question for them:
https://github.com/HumanDynamics/openbadge/issues/115#issue-382709017

bhimasta, 2018-11-21:
Considering we ordered to PCBWay using the same source of design, they didn't ask anything about this issue in my case. Is it weird, isn't it?

OrenLederman, 2018-11-21:
Patric - that is correct, no hole there. This large pad is part of the alternative battery holder we used to use in the past. Now the new battery holder is just soldered on top of it.

I kept the old design there as a backup since every change I make to the design triggers an order of another batch of prototypes and tests. It's quite common to have additional "stuff" in the hardware design that isn't used.

patcon, 2018-11-21:
Thanks Oren!

closed 2018-11-21 by patcon